arnold-benchmark/arnold #2: Any ETA for making the benchmark dataset available?

Jack-XHP, 2023-05-01:
Hi this is an exciting and impressive work for a photo-realistic robot task demonstration dataset.  I would like to try it out for my project, any ETA for releasing the dataset or the rendering code base?

nikepupu, 2023-05-02:
Hi Jack,
Thanks for your interest in our project. The code refactoring is taking longer than expected. We will be providing a docker image when we release everything, as the setup process is non-trivial. We will let you know when everything is ready.

chanhee-luke, 2023-06-20:
Hi @nikepupu. Is there any update on the progress? This dataset looks to be very exciting and I'm looking forward to using it!

huangjy-pku, 2023-08-12:
The release finally comes. Really sorry for such a delay.

closed 2023-08-12 by huangjy-pku