arnold-benchmark/arnold #5: Corrupted data

Reagan1311, 2023-08-28:
Hi, I got the following warnings when evaluating the "open drawer" task, and results are also worse compared to the original paper: 
`test: 25.21/31.09
; object: 1.94/6.45
; scene: 11.76/20.78
; state: 9.77/10.92
; any state: 15.13/21.85`

I tried to re-download the data, but the warnings remained. I wonder if there is something wrong with the data? Thanks.

[image]

nikepupu, 2023-08-28:
Hi Gen, 
Thanks for your interest in our work.
This error is expected.  This is a known issue with Isaac-Sim. 

https://forums.developer.nvidia.com/t/urdf-importer-visuals-has-corrupted-data-in-primvar-st-buffer-size-24-doesnt-match-expected-size-in-facevarying-primvars/218744/9

You can check the same issue also exists here in version 2022.2.0 :
https://github.com/NVIDIA-Omniverse/Orbit/issues/1

As far as I know, I don't think it will actually cause any issues in terms of experiments, but it will cause some issues when you try to debug something, as the terminal screens are overloaded with warnings. 

I am not sure about the timeline for when Nvidia will fix this issue though.

On the reproduced results, the physics simulation is not deterministic;  We recommend running the experiment at least 3 times.

Reagan1311, 2023-08-28:
Thanks for the prompt reply! Indeed, the variance of the experimental results are huge because of the small number of data in each test split and non-deterministic simulation. The difference in success rate of different evaluations in the same checkpoint can be 5~8%.

closed 2023-08-28 by Reagan1311