arnold-benchmark/arnold #6: Warning during evaluation on multi-task model

Reagan1311, 2023-09-02:
Hi there, 

I got the following warning every time when evaluating a multi-task model, and then the evaluation hangs forever. I wonder if this is due to the longer evaluation time in the multi-task setup, or some other errors encounterd in simulation? is there any solution for this? Thanks.

[image]

nikepupu, 2023-09-02:
Hi, Can you provide a log file?

There is an issue with multi-task evaluation. 
https://docs.omniverse.nvidia.com/isaacsim/latest/known_issues.html

As indicated here, when generating data, the replicator will cause some memory leak. When that happens, the VRAM will increase and eventually exceed the total VRAM and crash Isaac sim. 

There is also another possibility. It seems loading assets too often will cause issues in some system configurations. 
There are three workarounds (ranked in terms of complexity of implementation): 
1. Evaluating tasks one by one. 
2. Using lazy loading, basically checking if the next evaluation episode is using the same (scene, object) pairs, only delete when necessary. However, this may not solve the issue completely.
3. Load all assets and scenes into the USD, and put them very far away without simulating physics or anything (this is important; otherwise, there will be a significant slowdown). When the episode starts, move the objects, robots, and scenes in-use to the corresponding locations.

Reagan1311, 2023-09-03:
Really appreciate the provided workarounds! They are very helpful, and I think the first one should be enough to solve the issue.
I also attached my log files below. They looks fine, as the warnings are not written into the log files, so I also attached the screenshot of the warnings when the evaluation hung.
[eval.log](https://github.com/arnold-benchmark/arnold/files/12506360/eval.log)
[eval_wo_gt.log](https://github.com/arnold-benchmark/arnold/files/12506364/eval_wo_gt.log)
[image]
[image]

closed 2023-09-10 by Reagan1311