arnold-benchmark/arnold #7: IsaacSim Not Loading in Docker

helen9975, 2023-09-11:
Hi, thanks for this awesome work!

When trying to run the initial getting started task within the docker container, 
`/isaac-sim/python.sh /isaac-sim/standalone_examples/api/omni.isaac.franka/pick_place.py`

It seems that the IsaacSim never loads and just hangs, see below:

[image]

Any idea how to solve this?

Thanks!

nikepupu, 2023-09-11:
Hi Helen,
Thanks for your interest in our work. 
There are three possibilities:
1. The driver version does not match our descriptions: (525.60.11) for Linux
2. The OS version does not match our descriptions: I only managed to get this to work on Ubuntu 18.04 and 20.04. The code base does not work on 22.04
3. can you make sure nvidia_icd.json and nvidia_layers.json are mapped into the correct location? If they are empty, there is something wrong with the settings.

helen9975, 2023-09-11:
The specs I am using are as follows:
* 530.30.02
* CUDA Version 12.01
* Ubuntu 20.04

Both nvidia_icd.json and nvidia_layers.json are mapped to the correct location. However, nvidia_layers.json is empty. Is there some settings that I should adjust?

nikepupu, 2023-09-11:
There are two issues,

Can you install driver version: 525.60.11? 

Can you check the location of the file: nvidia_layers.json on the host machine? It's not supposed to be empty.

helen9975, 2023-09-11:
On the host machine, nvidia_layers.json is an empty directory, see below

[image]

nikepupu, 2023-09-11:
uhhh, that means there is something wrong with the driver installation 

Can you also check this location: /etc/vulkan/implicit_layer.d/nvidia_layers.json?

You might need to manually edit the Vagrantfile

helen9975, 2023-09-11:
/etc/vulkan/implicit_layer.d is empty for me

nikepupu, 2023-09-12:
I would recommend reinstalling the driver using run files.

You can refer to here: https://docs.omniverse.nvidia.com/isaacsim/latest/install_container.html

helen9975, 2023-09-12:
I have reinstalled the driver, now it is 525.60.11. nvidia_icd.json and nvidia_layers.json are also non-empty. However, now there is another libGL related error when I tried running pick_place.py: 
[image]

helen9975, 2023-09-12:
For reference, a fix for this error for me was to run

closed 2023-09-12 by helen9975

wusuoweima, 2024-01-12:
where to download 525.60.11? I can not find it.  Could you provide the file or link ?
@nikepupu @helen9975