arnold-benchmark/arnold #8: Demonstration Generator

amirhossein-razlighi, 2023-09-13:
Hi! When is the demonstration generator code will be published? We need that for re-generating the dataset with custom views using Isaac Sim.
Thanks.

nikepupu, 2023-09-13:
Hi, 
Thanks for your interest in our project.
I will get it done as soon as I have more time. The demonstration generation depends on a lot of stuff that is now deprecated. 

For now, a strategy is to replay the provided actions in the gt frames.

amirhossein-razlighi, 2023-09-13:
Thanks for your response. I checked out the eval code so that I can replay the provided actions in gt frame, but the problem is it's not much customisable so that we can get more information from the environment and add them to the dataset.
looking forward to the main code you used for generating the dataset provided.

nikepupu, 2023-09-13:
Can you elaborate on what information you need? Feel free to send me an email.

amirhossein-razlighi, 2023-09-13:
thanks! how can I contact you? I can not find the email address on your profile

nikepupu, 2023-09-13:
You can use the email contacts listed on the paper.

closed 2024-04-01 by huangjy-pku