arnold-benchmark/arnold #9: Two problematic data samples

Reagan1311, 2023-09-16:
Hi, I found two following problematic data samples during evaluation.

close_cabinet/novel_scene/Xiaofeng-close_cabinet-6-0-0.5-0.0-18-Wed_Jan_18_23:55:06_2023.npz
pour_water/novel_state/AWS-pour_water-2-0-0-75-5-Tue_Feb__7_10:24:42_2023.npz

The first data returned the error of `ValueError: Found zero norm quaternions in 'quat'` from [this line](https://github.com/arnold-benchmark/arnold/blob/ee6c4840517ef46f159971b611bf3df998db261b/eval.py#L160).
The second one has an inconsistent length of data. It returned the `IndexError: index 4 is out of bounds for axis 0 with size 4` from [this line](https://github.com/arnold-benchmark/arnold/blob/ee6c4840517ef46f159971b611bf3df998db261b/eval.py#L142).

Could you please check these data? Thanks!

nikepupu, 2023-09-17:
Hi,
Just to clarify, does this happen during replaying keyframe actions?

Reagan1311, 2023-09-18:
Yes, this also happened with `use_gt=[1,1]`, as the errors are from the gt data not the model prediction.

huangjy-pku, 2023-09-19:
Hello, @Reagan1311, you are right. There are flaws in these two data pieces. We will update the data. Thanks!

Reagan1311, 2023-09-19:
Hi, I got one follow-up question. It seems the evaluation process didn't use the predicted gripper open state. Instead, the gt gripper state from config file [here](https://github.com/arnold-benchmark/arnold/blob/ee6c4840517ef46f159971b611bf3df998db261b/configs/default.yaml#L88) is used. In this case, the prediction of the gripper open state is never utilized. May I ask why it was handled like this? Thanks!

nikepupu, 2023-09-20:
There are two reasons.
1. for a fair comparison with 6D cliport
2. In our two-phase execution, basically, the gripper is only open in the first stage.

closed 2023-09-24 by nikepupu