arnold-benchmark/arnold #11: Isaac sim issue

abdotaarek, 2023-10-17:
Hello,
When i try to install isaac sim 2022.1.1 from nividia omniverse it shows this error :
Error occurred during installation of Isaac Sim: X Error of failed request:  GLXBadFBConfig
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  0 ()
  Serial number of failed request:  201
  Current serial number in output stream:  201
2023-10-16 15:30:29 [839ms] [Error] [omni.core.ITypeFactory] ITypeFactory must not be destructed post-main. plugins will not be unloaded.
/media/local/atarek/isaac_sim-2022.1.1/omni.isaac.sim.post.install.sh: line 17: 38651 Segmentation fault      ${SCRIPT_DIR}/omni.isaac.sim.warmup.sh

nikepupu, 2023-10-17:
Maybe this helps: 

https://forums.developer.nvidia.com/t/failed-installation-of-isaac-sim-glxbadfbconfig/222584/2

abdotaarek, 2023-10-25:
Yes it helped, thank you so much

closed 2023-10-25 by abdotaarek