arnold-benchmark/arnold #14: nvidia-driver-525.60.11 has security vulnerabilities

teason30, 2023-11-21:
NVIDIA driver version 525.60.11 is quite old and has security vulnerabilities.  Can other versions of the 525 driver be used?  It looks as though anything prior to 525.125.06 is vulnerable.

https://nvidia.custhelp.com/app/answers/detail/a_id/5468

atharva7am, 2024-01-11:
Any updates on this? I'm working on a headless server where 525.60.11 shall not be installed due to vulnerability issues.
Docker support for the latest drivers is much requested. @nikepupu @huangjy-pku

nikepupu, 2024-01-11:
Maybe you can try to play with the Dockerfile? Community support is much appreciated for this issues at the moment. I apologize for this.  I am not really an expert on isaac sim - nvidia driver compatability issues.  I actually tried a bunch of versions. The one I mentioned is the only driver version I found to be functional around the time of release along with the docker file.

nikepupu, 2024-02-21:
I can confirm 535.113.01 also works.

closed 2024-04-01 by huangjy-pku