arnold-benchmark/arnold #15: Training Issue

JasonQSY, 2023-12-18:
Thanks for the great work! I'm trying to reproducing the peract training on ARNOLD. Up till now, I have installed the environment using docker and the visualization code works for me.

However, when I try to start the training, it throws a weird socket error and I'm not sure what's wrong. Do you have any hints?

```
/isaac-sim/python.sh train_peract.py task=pickup_object model=peract lang_encoder=clip mode=train batch_size=8 steps=100000
train_peract.py:151: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path='./configs', config_name='default')
/isaac-sim/kit/python/lib/python3.7/site-packages/hydra/_internal/hydra.py:127: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  configure_logging=with_log_configuration,
Loaded [86, 84, 70, 80, 100, 119, 36, 48] demos
Error executing job with overrides: ['task=pickup_object', 'model=peract', 'lang_encoder=clip', 'mode=train', 'batch_size=8', 'steps=100000']
Traceback (most recent call last):
  File "/isaac-sim/kit/python/lib/python3.7/urllib/request.py", line 1350, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/isaac-sim/kit/python/lib/python3.7/http/client.py", line 1281, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/isaac-sim/kit/python/lib/python3.7/http/client.py", line 1327, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/isaac-sim/kit/python/lib/python3.7/http/client.py", line 1276, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/isaac-sim/kit/python/lib/python3.7/http/client.py", line 1036, in _send_output
    self.send(msg)
  File "/isaac-sim/kit/python/lib/python3.7/http/client.py", line 976, in send
    self.connect()
  File "/isaac-sim/kit/python/lib/python3.7/http/client.py", line 1443, in connect
    super().connect()
  File "/isaac-sim/kit/python/lib/python3.7/http/client.py", line 948, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/isaac-sim/kit/python/lib/python3.7/socket.py", line 707, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "/isaac-sim/kit/python/lib/python3.7/socket.py", line 752, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Temporary failure in name resolution

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "train_peract.py", line 165, in main
    lang_encoder = create_lang_encoder(cfg, device)
  File "train_peract.py", line 25, in create_lang_encoder
    return CLIP_encoder(device)
  File "/root/arnold/peract/agent.py", line 56, in __init__
    self.model, preprocess = clip.load("RN50", device=device, jit=False)
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/clip/clip.py", line 64, in load
    model_path = _download(_MODELS[name])
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/clip/clip.py", line 40, in _download
    with urllib.request.urlopen(url) as source, open(download_target, "wb") as output:
  File "/isaac-sim/kit/python/lib/python3.7/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/isaac-sim/kit/python/lib/python3.7/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/isaac-sim/kit/python/lib/python3.7/urllib/request.py", line 543, in _open
    '_open', req)
  File "/isaac-sim/kit/python/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/isaac-sim/kit/python/lib/python3.7/urllib/request.py", line 1393, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "/isaac-sim/kit/python/lib/python3.7/urllib/request.py", line 1352, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno -3] Temporary failure in name resolution>

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
There was an error running python
```

huangjy-pku, 2023-12-18:
Hello, Shengyi! Seems like a problem when downloading pre-trained CLIP weights. Make sure your network can access CLIP weights. An alternative is to download the weights manually and place them in `~/.cache/clip/`.

closed 2024-04-01 by huangjy-pku

charlottejruby, 2024-09-11:
Hello, May I ask where I can download the pretrained weights?
Thank you.

JasonQSY, 2024-09-11:
Yeah I can confirm I was missing clip weights. I believe it's this one https://github.com/openai/CLIP But I eventually copy it from another server, so I cannot guarantee it is correct...

huangjy-pku, 2024-09-12:
One solution is to initialize the pretrained CLIP model somewhere you have network access (e.g., your laptop), and then copy the weights in `~/.cache/clip/` to your server if there is no network access.