arnold-benchmark/arnold #17: GLFW initialization failed.

wusuoweima, 2024-01-05:
我在服务器中的子账户 的docker 里运行 /isaac-sim/python.sh /isaac-sim/standalone_examples/api/omni.isaac.franka/pick_place.py

出现了下面的错误，是因为我的服务器没有界面吗？
事先也运行了  sudo apt install -y docker.io vagrant freerdp2-x11

```
1. 2024-01-04 08:44:35 [23,045ms] [Error] [carb.windowing-glfw.plugin] GLFW initialization failed.
2. 2024-01-04 08:44:35 [23,045ms] [Error] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.3]) (impl: carb.windowing-glfw.plugin)
3. 2024-01-04 08:44:35 [23,045ms] [Error] [carb.windowing-glfw.plugin] GLFW initialization failed.
4. 2024-01-04 08:44:35 [23,045ms] [Error] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.3]) (impl: carb.windowing-glfw.plugin)
5. 2024-01-04 08:44:35 [23,046ms] [Error] [omni.appwindow.plugin] Failed to acquire IWindowing interface
```
@nikepupu
@huangjy-pku

nikepupu, 2024-01-05:
Hi you can modify the code to set headless to True.

wusuoweima, 2024-01-09:
我换到了个人台式机上配置，运行 /isaac-sim/python.sh /isaac-sim/standalone_examples/api/omni.isaac.franka/pick_place.py 后一直卡在下面的位置
[image]

nikepupu, 2024-01-10:
please make sure the driver is installed properly as : https://github.com/arnold-benchmark/arnold/issues/7

closed 2024-04-01 by huangjy-pku