arnold-benchmark/arnold #18: editing ground truth actions

abdotaarek, 2024-01-23:


abdotaarek, 2024-01-23:
hello, is it possible to change the ground truth actions

huangjy-pku, 2024-01-24:
Of course, you can assign `gt_actions` in `eval.py` on your own.

abdotaarek, 2024-01-24:
Also, one more question, how can i evaluate only one episode that i choose from the demonstration npz files.

huangjy-pku, 2024-01-25:
Your code is ok, I think. To evaluate on a specific demonstration, you can add some operations in `load_data` function. For example:
```python
for npz_path in demo_path:
    if file_name not in npz_path:   # add
        continue   # add
    data.append(np.load(npz_path, allow_pickle=True))
    frames.append(npz_path)
```

abdotaarek, 2024-01-26:
Thank you, appreciate the fast response.

abdotaarek, 2024-01-27:
Hi, i gave the peract model the customized gt as you said in the eval script, i gave it the position, orientation, and the gripper state.
But the task failed, should the Gt actions be more than these 3 informations?

nikepupu, 2024-01-27:
How do you provide position, orientation, and the gripper state? Peract does not have 100 % success rate . Please refer to the paper.

huangjy-pku, 2024-01-27:
Why would you overwrite the gt actions? We have provided them in the demonstration file. If you want to overwrite, make sure the customized action is accurate.

closed 2024-01-27 by abdotaarek