arnold-benchmark/arnold #20: Stuck at Sanity Check

qwerty863, 2024-04-03:
Hi, I'm working on the docker setup and it looks like the sanity check code does not proceed.

This is the code that the document provides.
```
/isaac-sim/python.sh /isaac-sim/standalone_examples/api/omni.isaac.franka/pick_place.py
```

After running this, the code does not proceed anymore as below.
```
root@ubuntu:~# /isaac-sim/python.sh /isaac-sim/standalone_examples/api/omni.isaac.franka/pick_place.py
Starting kit application with the fillowing args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--allow-root']
Passing the following args to the base kit application:  []
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.1/kit_20240403_095716.log
2024-04-03 09:57:16 [2ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar/config'
[0.111s] [ext: omni.stats-0.0.0] startup
[0.130s] [ext: omni.gpu_foundation-0.0.0] startup
2024-04-03 09:57:16 [128ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.5], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
[0.136s] [ext: carb.windowing.plugins-1.0.0] startup
[0.151s] [ext: omni.assets.plugins-0.0.0] startup
[0.152s] [ext: omni.kit.renderer.init-0.0.0] startup
```

nikepupu, 2024-04-03:
Hi Can you verify vagrant path is set up correctly as indicated in 
https://arnold-docs.readthedocs.io/en/latest/tutorial/setup/index.html
part 4.

qwerty863, 2024-04-04:
Hi, yes, I double-checked the vagrant path and found that the two paths are valid.

nikepupu, 2024-04-04:
Hi, What is your driver version?

nikepupu, 2024-04-04:
[image]

nikepupu, 2024-04-04:
I tested extensively and a lot of the 525 drivers do not work at all.

qwerty863, 2024-04-04:
I see, then it looks like I need to switch to the exact 525.60.11 or 535.113.01.
Thanks for your help!

qwerty863, 2024-04-06:
Hi, @nikepupu

I tried 525.60.11 but it is still stuck at the beginning of the run.
I used the exact driver version (525.60.11) and ensured valid vagrant paths but it's still not working.
Do you have any thoughts on some possible reasons?

Thanks.

nikepupu, 2024-04-07:
can you try to modify the file to set headless to True?

qwerty863, 2024-04-07:
I tried, but the problem still remains.

nikepupu, 2024-04-07:
what OS version are you using?

qwerty863, 2024-04-07:
I'm using Ubuntu 20.04

nikepupu, 2024-04-07:
maybe you can refer to this: https://forums.developer.nvidia.com/t/isaac-sim-hangs-on-ext-omni-kit-renderer-init-0-0-0-startup/246984/6

nikepupu, 2024-04-07:
uninstall driver
delete:
/usr/share/vulkan
/etc/vulkan

install driver

qwerty863, 2024-04-07:
Thanks! I'll try this.

nikepupu, 2024-04-11:
BTW what GPU are you using?

closed 2024-05-10 by huangjy-pku