arnold-benchmark/arnold #22: No module named 'carb'

xiaohao9999, 2024-04-06:
Hi,

When I complete the training and run the ckpt_selection file, the following error occurs:

```
Traceback (most recent call last):
  File "eval.py", line 19, in <module>
    from environment.runner_utils import get_simulation
  File "/data1/ws/arnold/environment/runner_utils.py", line 4, in <module>
    import carb
ModuleNotFoundError: No module named 'carb'
```

Can you guide me on how to do it?

xiaohao9999, 2024-04-07:
It can only be evaluate based on docker installation, right?

nikepupu, 2024-04-07:
Hi, Carb is a module used by nvidia isaac sim. Did you have isaac sim installed?

xiaohao9999, 2024-04-08:
Hi,

I noticed https://github.com/arnold-benchmark/arnold/issues/14#issuecomment-1955647785. Can you install isaac sim 2022.1.1 under the 535.113.01?

My server driver version is 530.41.03, But I always can’t install  isaac sim 2022.1.1. Error occurred:
```
/data/.loacl/share/ov/pkg/isaac_sim-2022.1.1/omni.isaac.sim.post.install.sh: line 17: 41841 Segmentation fault      (core dumped) ${SCRIPT_DIR}/omni.isaac.sim.warmup.sh
```
Can you guide me on how to do it?

nikepupu, 2024-04-08:
On my Ubuntu 20.04, with driver version 535.113.01. The installation works. I can install it from the launcher.

nikepupu, 2024-04-08:
maybe refer to here: https://github.com/arnold-benchmark/arnold/issues/20

xiaohao9999, 2024-04-11:
Sorry to bother you again. When I execute `ckpt_selection, some errors occur:

[image]

nikepupu, 2024-04-11:
It seems that assets are not placed in the correct location.

closed 2024-04-11 by xiaohao9999