arnold-benchmark/arnold #29: Issues for submission of final results

ljy1234-special, 2024-05-11:
Hello. I'm sorry to bother you again. I have submitted the baseline after entering the submit command 'evalai challenge 2266 phase 4501 submit --file submission_test.zip --large ---public' in conda environment. But the baseline I tested locally is not 0. What is the reason for this?
[image]

nikepupu, 2024-05-11:
Which split did you run? Did you run the dev split but submit to the test split?

nikepupu, 2024-05-11:
please make sure to use the challenge branch.

ljy1234-special, 2024-05-11:
I'm sure we ran and submitted the test split.The three steps are as follows:
[image]
[image]
[image]

Also, we do use main branch for` training` and `evaluation`. So, is it feasible to use `the results of main brach training` for testing in `challenge branch`? Or do we have to retrain in `challenge branch`?

nikepupu, 2024-05-11:
You don't have to retrain. Can you send me an email with the submission ID?

ljy1234-special, 2024-05-11:
Okay, I'm sending it.

nikepupu, 2024-05-11:
I found something very interesting: the file name will differ when using non-English operating systems. I am working on a fix now.

nikepupu, 2024-05-11:
It's now fixed. I have rerun the submission; you should be able to see the updated results. 

Thank you for bringing this up!

closed 2024-05-12 by nikepupu