arnold-benchmark/arnold #30: Problems with the results of the experiment

renyu2016, 2024-05-12:
Hi,
Thank you for release such a excellent work !
After setup the docker-based environment and preparing the data, I try to visualize the demonstration as Tutorial
`/isaac-sim/python.sh eval.py task=pour_water mode=eval use_gt=[1,1] visualize=1`
I can see the the window of isaac sim is ready but i got following error

> 
> [4.606s] app ready
> [6.710s] Simulation App Startup Complete
> 2024-05-12 04:10:20 [6,767ms] [Warning] [carb.flatcache.plugin] Type tag does not have a corresponding USD type
> eval.py:82: UserWarning: 
> The version_base parameter is not specified.
> Please specify a compatability version level, or None.
> Will assume defaults for version 1.1
>   @hydra.main(config_path='./configs', config_name='default')
> /isaac-sim/kit/python/lib/python3.7/site-packages/hydra/_internal/hydra.py:127: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
> See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
>   configure_logging=with_log_configuration,
> [2024-05-12 04:10:21,038][__main__][INFO] - Evaluating pour_water test
> Error executing job with overrides: ['task=pour_water', 'mode=eval', 'use_gt=[1,1]', 'visualize=1']
> Traceback (most recent call last):
>   File "eval.py", line 170, in main
>     assert gt_actions[2] is not None, "Use second gt action but it is missing"
> AssertionError: Use second gt action but it is missing
> 
> Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
> /isaac-sim/python.sh: line 40: 15167 Segmentation fault      (core dumped) $python_exe "$@" $args
> There was an error running python

[image]

Could you please give me any advice for this issue? 
Thanks in advance!!

nikepupu, 2024-05-12:
Hi, for non-training data in the challenge, we did not provide it for the 2nd stage. If you want to evaluate the model, please set 'use_gt=[0,0]'

renyu2016, 2024-05-12:
Thanks for your response!
After changing data , I can load environment and visualize the demonstration. But it seems like there are some others problems when loading the environment in isaac sim:

[image]

and the images shown in isaac sim window is a little dark and noisy

[image]

 Is this image rendering correct?

nikepupu, 2024-05-12:
the resolution by default is 128 by 128. If you want a higher resolution image,  you can modify it as well.

renyu2016, 2024-05-12:
So,  the error reported by isaac sim could be ignored, is that right?

nikepupu, 2024-05-12:
Most of the errors can be ignored. It seems there is an issue with the skylight, though.

renyu2016, 2024-05-12:
Yes, it's pretty wired. I can download the mdl file using `wget http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Skies/Sky_Elements/materials/procedural/ClearSky.mdl` smoothly, but the isaac sim cannot get it by it self. So the light cannot be loaded and the entire environment is looked very dark.

 I create and save a distant light using isaac sim by myself and replace the skylight with it in the arnold. Is that ok for training and testing?

nikepupu, 2024-05-12:
You don't need lightings for training since we have saved the dataset already. For evaluation, feel free to explore what works best for you.

renyu2016, 2024-05-13:
I got it! Thank you very much!

closed 2024-05-13 by renyu2016