arnold-benchmark/arnold #32: usd error when running training example

GuoPingPan, 2024-05-20:
Traceback (most recent call last):
  File "train_peract.py", line 18, in <module>
    from dataset import ArnoldDataset, ArnoldMultiTaskDataset, InstructionEmbedding
  File "/home/a6000/arnold/dataset.py", line 9, in <module>
    from utils.env import CAMERAS
  File "/home/a6000/arnold/utils/env.py", line 6, in <module>
    from .transforms import create_pcd_hardcode, quat_diff_rad, get_pose_relat, get_pose_world
  File "/home/a6000/arnold/utils/transforms.py", line 3, in <module>
    from pxr import Gf
  File "/home/a6000/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/extscore/omni.usd.libs/pxr/Gf/__init__.py", line 31, in <module>
    from pxr import Tf
  File "/home/a6000/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/extscore/omni.usd.libs/pxr/Tf/__init__.py", line 85, in <module>
    from . import _tf
ImportError: /home/a6000/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/extscore/omni.usd.libs/pxr/Tf/_tf.so: undefined symbol: _ZTIN32pxrInternal_v0_20__pxrReserved__8TfNotice14_DelivererBaseE

GuoPingPan, 2024-05-20:
if `pip install usd-core` will encounter the error mentioned in https://github.com/arnold-benchmark/arnold/issues/24

nikepupu, 2024-05-20:
how about uninstalling usd-core?

GuoPingPan, 2024-05-20:
I have not install usd-core.

GuoPingPan, 2024-05-20:
```
python train_peract.py task=pickup_object model=peract lang_encoder=clip mode=train batch_size=8 steps=100000
```
will output
```
Traceback (most recent call last):
  File "train_peract.py", line 18, in <module>
    from dataset import ArnoldDataset, ArnoldMultiTaskDataset, InstructionEmbedding
  File "/home/a6000/arnold/dataset.py", line 9, in <module>
    from utils.env import CAMERAS
  File "/home/a6000/arnold/utils/env.py", line 6, in <module>
    from .transforms import create_pcd_hardcode, quat_diff_rad, get_pose_relat, get_pose_world
  File "/home/a6000/arnold/utils/transforms.py", line 3, in <module>
    from pxr import Gf
  File "/home/a6000/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/extscore/omni.usd.libs/pxr/Gf/__init__.py", line 31, in <module>
    from pxr import Tf
  File "/home/a6000/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/extscore/omni.usd.libs/pxr/Tf/__init__.py", line 85, in <module>
    from . import _tf
ImportError: /home/a6000/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/extscore/omni.usd.libs/pxr/Tf/_tf.so: undefined symbol: _ZTIN32pxrInternal_v0_20__pxrReserved__8TfNotice14_DelivererBaseE

```

nikepupu, 2024-05-20:
Can you try the command pip uninstall usd-core first?

GuoPingPan, 2024-05-20:
[image]

GuoPingPan, 2024-05-20:
@nikepupu Is there any solution?

huangjy-pku, 2024-05-20:
Hello, seems you are taking the `conda` approach instead of `docker` approach? If so, make sure you have followed the [guidelines](https://arnold-docs.readthedocs.io/en/latest/tutorial/setup/index.html#conda-based). In regard to your issue, I suspect there may be an improper activation of `Issac Sim` env:
```
source ${Isaac_Sim_Root}/setup_conda_env.sh
```

GuoPingPan, 2024-05-20:
I am strictly follow the conda-based step up which can pass `sanity check`.
but encounter the above problem when running  
```
python train_peract.py task=pickup_object model=peract lang_encoder=clip mode=train batch_size=8 steps=100000
```

closed 2024-05-31 by nikepupu