arnold-benchmark/arnold #33: Problems encountered when close_cabinet

windboy6, 2024-05-24:
Hello, thank you for sharing such great work. I use `accelerate` for distributed training. Then when close drawer, the results are consistent with the accuracy of the paper. Bur when I reproduced close and open cabinet, I found that the results was not as reported in the paper.
`"peract_open_cabinet_rgb_clip_150000.pth": {
    "open_cabinet": {
      "score": 0.05
    }
  },
  "peract_open_cabinet_rgb_clip_160000.pth": {
    "open_cabinet": {
      "score": 0.0
    }
  },
  "peract_open_cabinet_rgb_clip_170000.pth": {
    "open_cabinet": {
      "score": 0.0
    }
  }`
  
<details>
  <summary>ckpt_selection.log</summary>
  
[2024-05-24 09:44:45,018][__main__][INFO] - Evaluating peract_open_cabinet_rgb_clip_170000.pth open_cabinet
[2024-05-24 09:44:47,627][__main__][INFO] - Instruction: slide the  closet half
[2024-05-24 09:44:47,627][__main__][INFO] - Ground truth action:
[2024-05-24 09:44:47,628][__main__][INFO] - trans=[-811.65883362   51.1315112   -75.67004328], orient(euler XYZ)=[-179.98908622   -1.61341236   89.96888096], gripper_open=True
[2024-05-24 09:44:47,628][__main__][INFO] - trans=[-811.84835783   51.09921275  -84.6938762 ], orient(euler XYZ)=[179.87591455  -1.46350374  90.3711248 ], gripper_open=False
[2024-05-24 09:44:47,628][__main__][INFO] - trans=[-821.75774595   51.03947223  -59.14556211], orient(euler XYZ)=[179.74846856  39.47749717  91.56142256], gripper_open=False
[2024-05-24 09:44:47,747][__main__][INFO] - Prediction action 0: trans=[-782.80768497   45.99299526  -59.95301794], orient(euler XYZ)=[180.  25.  90.]
[2024-05-24 09:44:47,747][tasks.open_cabinet][INFO] - tick: 0
[2024-05-24 09:44:47,959][tasks.open_cabinet][INFO] - tick: 120
[2024-05-24 09:44:47,970][tasks.open_cabinet][INFO] - enter stage 1
[2024-05-24 09:44:48,189][tasks.open_cabinet][INFO] - enter stage 2
[2024-05-24 09:44:48,393][__main__][INFO] - Prediction action 1: trans=[-748.78768355   34.65299892  -61.21301841], orient(euler XYZ)=[180.  45.  90.]
[2024-05-24 09:44:48,407][tasks.open_cabinet][INFO] - tick: 240
[2024-05-24 09:44:48,585][tasks.open_cabinet][INFO] - tick: 360
[2024-05-24 09:44:48,761][tasks.open_cabinet][INFO] - tick: 480
[2024-05-24 09:44:48,940][tasks.open_cabinet][INFO] - tick: 600
[2024-05-24 09:44:49,118][tasks.open_cabinet][INFO] - tick: 720
[2024-05-24 09:44:49,304][tasks.open_cabinet][INFO] - tick: 840
[2024-05-24 09:44:49,491][tasks.open_cabinet][INFO] - tick: 960
[2024-05-24 09:44:49,677][tasks.open_cabinet][INFO] - tick: 1080
[2024-05-24 09:44:49,865][tasks.open_cabinet][INFO] - tick: 1200
[2024-05-24 09:44:50,048][tasks.open_cabinet][INFO] - tick: 1320
[2024-05-24 09:44:50,224][tasks.open_cabinet][INFO] - tick: 1440
[2024-05-24 09:44:50,404][tasks.open_cabinet][INFO] - tick: 1560
[2024-05-24 09:44:50,578][tasks.open_cabinet][INFO] - tick: 1680
[2024-05-24 09:44:50,754][tasks.open_cabinet][INFO] - tick: 1800
[2024-05-24 09:44:50,929][tasks.open_cabinet][INFO] - tick: 1920
[2024-05-24 09:44:51,106][tasks.open_cabinet][INFO] - tick: 2040
[2024-05-24 09:44:51,276][tasks.open_cabinet][INFO] - tick: 2160
[2024-05-24 09:44:51,450][tasks.open_cabinet][INFO] - tick: 2280
[2024-05-24 09:44:51,626][tasks.open_cabinet][INFO] - tick: 2400
[2024-05-24 09:44:51,799][__main__][INFO] - correct: 0 | total: 1 | remaining: 19

[2024-05-24 09:44:54,987][__main__][INFO] - Instruction: slide the  cupboard two quarters
[2024-05-24 09:44:54,987][__main__][INFO] - Ground truth action:
[2024-05-24 09:44:54,987][__main__][INFO] - trans=[-511.28500086   64.37152509   76.0020443 ], orient(euler XYZ)=[-1.79997019e+02 -6.46351537e+01 -1.31485178e-02], gripper_open=True
[2024-05-24 09:44:54,988][__main__][INFO] - trans=[-519.39950397   64.51322792   72.13756245], orient(euler XYZ)=[-179.68593144  -64.63394792    0.2400287 ], gripper_open=False
[2024-05-24 09:44:54,988][__main__][INFO] - trans=[-511.6264002    63.35352263   75.87539553], orient(euler XYZ)=[176.59950857 -43.77034213   0.73581477], gripper_open=False
[2024-05-24 09:44:55,109][__main__][INFO] - Prediction action 0: trans=[-475.78639755   62.37299252   75.7029414 ], orient(euler XYZ)=[180.   0.   0.]
[2024-05-24 09:44:55,109][tasks.open_cabinet][INFO] - tick: 0
[2024-05-24 09:44:55,312][tasks.open_cabinet][INFO] - tick: 120
[2024-05-24 09:44:55,498][tasks.open_cabinet][INFO] - tick: 240
[2024-05-24 09:44:55,682][tasks.open_cabinet][INFO] - tick: 360
[2024-05-24 09:44:55,868][tasks.open_cabinet][INFO] - tick: 480
[2024-05-24 09:44:55,958][tasks.open_cabinet][INFO] - enter stage 1
[2024-05-24 09:44:56,053][tasks.open_cabinet][INFO] - tick: 600
[2024-05-24 09:44:56,237][tasks.open_cabinet][INFO] - tick: 720
[2024-05-24 09:44:56,409][tasks.open_cabinet][INFO] - tick: 840
[2024-05-24 09:44:56,546][tasks.open_cabinet][INFO] - tick: 960
[2024-05-24 09:44:56,678][tasks.open_cabinet][INFO] - tick: 1080
[... 531 lines cut ...]
[2024-05-24 09:47:19,875][tasks.open_cabinet][INFO] - tick: 360
[2024-05-24 09:47:20,011][tasks.open_cabinet][INFO] - enter stage 1
[2024-05-24 09:47:20,109][tasks.open_cabinet][INFO] - tick: 480
[2024-05-24 09:47:20,326][tasks.open_cabinet][INFO] - tick: 600
[2024-05-24 09:47:20,542][tasks.open_cabinet][INFO] - tick: 720
[2024-05-24 09:47:20,757][tasks.open_cabinet][INFO] - tick: 840
[2024-05-24 09:47:20,971][tasks.open_cabinet][INFO] - tick: 960
[2024-05-24 09:47:21,186][tasks.open_cabinet][INFO] - tick: 1080
[2024-05-24 09:47:21,401][tasks.open_cabinet][INFO] - tick: 1200
[2024-05-24 09:47:21,616][tasks.open_cabinet][INFO] - tick: 1320
[2024-05-24 09:47:21,829][tasks.open_cabinet][INFO] - tick: 1440
[2024-05-24 09:47:22,043][tasks.open_cabinet][INFO] - tick: 1560
[2024-05-24 09:47:22,256][tasks.open_cabinet][INFO] - tick: 1680
[2024-05-24 09:47:22,468][tasks.open_cabinet][INFO] - tick: 1800
[2024-05-24 09:47:22,684][tasks.open_cabinet][INFO] - tick: 1920
[2024-05-24 09:47:22,897][tasks.open_cabinet][INFO] - tick: 2040
[2024-05-24 09:47:23,109][tasks.open_cabinet][INFO] - tick: 2160
[2024-05-24 09:47:23,327][tasks.open_cabinet][INFO] - tick: 2280
[2024-05-24 09:47:23,494][tasks.open_cabinet][INFO] - tick: 2400
[2024-05-24 09:47:23,651][__main__][INFO] - correct: 0 | total: 20 | remaining: 0

[2024-05-24 09:47:23,651][__main__][INFO] - peract_open_cabinet_rgb_clip_170000.pth open_cabinet: 0.00  
</details>

[ckpt_selection.log.txt](https://github.com/arnold-benchmark/arnold/files/15426447/ckpt_selection.log.txt)

nikepupu, 2024-05-24:
Hi, the released test set is for the challenge, which is a much smaller split. We will release the full dataset after the challenge.

windboy6, 2024-05-25:
Ok, I'll test again then. Thanks.

closed 2024-05-25 by windboy6

ljy1234-special, 2024-08-19:
Hi, have you released the full dataset yet? I see that the dataset revision time on Google Drive is still the same as before. If I wish to reproduce the results from the paper, which folder of data should I use?
[image]

ljy1234-special, 2024-08-19:
Thank you！

> Hi,
> 
> you can use the data inside the tasks folder. I moved it away during the challenge.