arnold-benchmark/arnold #36: docker pull failed

Student-of-Holmes, 2024-07-24:
Hello, 
thank you for providing such a good benchmark. I'm trying to install the benchmark environment via docker. However, the image can't be pulled. Here is the detailed problem.
(base) user@user-System-Product-Name:~/project/Arnold/arnold/workspace$ sudo docker build -f Dockerfile -t "arnold" .
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  9.216kB
Step 1/43 : ARG DEBIAN_FRONTEND=noninteractive
Step 2/43 : ARG BASE_DIST=ubuntu20.04
Step 3/43 : ARG CUDA_VERSION=11.4.2
Step 4/43 : ARG ISAACSIM_VERSION=2022.1.1
Step 5/43 : FROM nvcr.io/nvidia/isaac-sim:${ISAACSIM_VERSION} as isaac-sim
 ---> f7b293669041
Step 6/43 : FROM nvidia/cudagl:${CUDA_VERSION}-base-${BASE_DIST}
11.4.2-base-ubuntu20.04: Pulling from nvidia/cudagl
8e5c1b329fe3: Retrying in 1 second 
8bd98d4761dc: Retrying in 1 second 
6ae8b24197d5: Retrying in 1 second 
598a82367886: Waiting 
4b017f335469: Waiting 
c4bb681693f5: Waiting 
db57642882ac: Waiting 
ed8638b45f6e: Waiting 
75b4743e7676: Waiting 
0b5aa3146c6f: Waiting 
c923aed82d3e: Waiting 
error pulling image configuration: download failed after attempts=6: dial tcp 43.226.16.8:443: i/o timeout

What should I do to tackle this problem?

nikepupu, 2024-07-24:
Hi are you sure you are using the provided Dockerfile? 

Everything seems to be working on my side, and I only have 28 steps, but it seems you have 43 steps. 
[stage-1 28/28]

In addition, it seems , we are using different docker versions: mine is Docker version 26.1.3, build b72abbb

Student-of-Holmes, 2024-07-25:
> Hi are you sure you are using the provided Dockerfile?
> 
> Everything seems to be working on my side, and I only have 28 steps, but it seems you have 43 steps. [stage-1 28/28]
> 
> In addition, it seems , we are using different docker versions: mine is Docker version 26.1.3, build b72abbb

I reinstall the docker in my pc. Now mine is Docker version 27.1.1, build 6312585
 Then I follow the instruction docs. The command is below (Notice that I have installed the nvidia container and logged into the NGC account)

```
git clone https://github.com/arnold-benchmark/arnold.git
cd arnold/workspace
docker build -f Dockerfile -t "arnold" .
```

I encounter a different problem. Here is the detail.
(base) user@user-System-Product-Name:~/project/Arnold/arnold/workspace$ sudo docker build -f Dockerfile -t "arnold" .
[+] Building 94.3s (3/3) FINISHED                            docker:default
 => [internal] load build definition from Dockerfile                   0.0s
 => => transferring dockerfile: 5.29kB                                 0.0s
 => WARN: FromAsCasing: 'as' and 'FROM' keywords' casing do not match  0.0s
 => ERROR [internal] load metadata for docker.io/nvidia/cudagl:11.4.  94.1s
 => [internal] load metadata for nvcr.io/nvidia/isaac-sim:2022.1.1     0.0s
  [internal] load metadata for docker.io/nvidia/cudagl:11.4.2-base-ubuntu20.04:
Dockerfile:9
   7 |     
   8 |     FROM nvcr.io/nvidia/isaac-sim:${ISAACSIM_VERSION} as isaac-sim
   9 | >>> FROM nvidia/cudagl:${CUDA_VERSION}-base-${BASE_DIST}
  10 |     
  11 |     RUN apt-get update && DEBIAN_FRONTEND="noninteractive" apt-get install -y neofetch unrar net-tools iputils-ping software-properties-common curl zip python3 python3-pip tmux git htop build-essential aria2 openssh-server xrdp xfce4* sudo vim libtf-dev && apt-get clean
ERROR: failed to solve: nvidia/cudagl:11.4.2-base-ubuntu20.04: failed to resolve source metadata for docker.io/nvidia/cudagl:11.4.2-base-ubuntu20.04: failed to copy: httpReadSeeker: failed open: unexpected status code https://alzgoonw.mirror.aliyuncs.com/v2/nvidia/cudagl/blobs/sha256:11ecffb024ea7ea2558d4ab0548359fb756efb1db0e570de575be02cd99518d3?ns=docker.io: 500 Internal Server Error - Server message: unknown: unknown error

nikepupu, 2024-07-25:
maybe the docker image is not hosted in aliyun as suggested here?

ERROR: failed to solve: nvidia/cudagl:11.4.2-base-ubuntu20.04: failed to resolve source metadata for docker.io/nvidia/cudagl:11.4.2-base-ubuntu20.04: failed to copy: httpReadSeeker: failed open: unexpected status code https://alzgoonw.mirror.aliyuncs.com/v2/nvidia/cudagl/blobs/sha256:11ecffb024ea7ea2558d4ab0548359fb756efb1db0e570de575be02cd99518d3?ns=docker.io: 500 Internal Server Error - Server message: unknown: unknown error

RenMing-Huang, 2024-08-21:
> Hello, thank you for providing such a good benchmark. I'm trying to install the benchmark environment via docker. However, the image can't be pulled. Here is the detailed problem. (base) [...]

Hi, I have the same problem, have you solved it？