arnold-benchmark/arnold #38: Docker build terminates with error

wpumacay, 2024-08-21:
Hi. Thanks for the great benchmark. I'm running into some issues when trying to setup using docker on my machine. It's an issue with the second step of the build process of the docker image. I used the commands as specified in the docs (`docker build -f Dockerfile -t "arnold" .`).I attach a screenshot of the error. Notice is the second step that fails. It says something went wrong with udisks2 and some gvfs related packages.

[image]

Some info that might be useful:
- `docker`: Docker version 27.1.2, build d01f264
- `OS`: Ubuntu22.04
- `driver`: 560.28.03

Note that all same steps work on another Ubuntu20.04 machine that I have access to. I'll try downgrading my driver to see if that could be a potential issue, but so far I got the same error. Thanks in advance for any help :+1:

nikepupu, 2024-08-21:
Hi,

In addition to driver, I am not sure if the isaac sim version we are using supports ubuntu 22.04.

wpumacay, 2024-08-21:
Hi @nikepupu . It seems it wasn't the driver, I still get the same error :sweat_smile: . You're right, it could be that it's not supported on 22.04. The other machine I have uses 20.04 and 474 for drivers and works just fine.

nikepupu, 2024-08-21:
Well, we probably need to do some upgrades by the end of the year.

wpumacay, 2024-08-23:
Hi @nikepupu . I ran into the same issue when installing on a separate ubuntu20.04 machine with nvidia driver 535. Docker also fails to build with the same issues. Could it be that my graphics card is not supported? I'm using an rtx3070.

nikepupu, 2024-08-25:
Can you match the minor version of the driver as well?

wpumacay, 2024-08-26:
Sure, uhmmm, which full version should I use, the recommended one (525.60.11) ?.

nikepupu, 2024-09-06:
[image]