arnold-benchmark/arnold #40: Problems with point cloud data for open_drawer tasks

ljy1234-special, 2024-08-30:
Hi, I am in the process of debugging the `open_drawer` task and I found that the pcd obtained from the `act2` phase in some datasets is `nan` as shown below. (The example shows '`Ziheng-open_drawer-7-1-0.0-0.25-0-Mon_Jan_30_14:01:07_2023.npz`'). May I ask what is the reason for this?How did you solve it?
[image]
[image]

nikepupu, 2024-08-31:
Is it possible some depth is inf?

ljy1234-special, 2024-08-31:
I found that the inf in the depth value can be eliminated after a clip operation, but the nan information remains.
[image]
[image]

huangjy-pku, 2024-09-02:
Hello, @ljy1234-special ! I checked the file you mentioned and indeed encountered the issue you mentioned. This is caused by some `nan` elements in the raw camera sensory data `camera_obs['depthLinear']` and `np.clip` cannot handle `nan` (though it can handle `inf`). More specifically, this issue only exists in the second camera `camera_obs = imgs[1]` in the third keyframe `imgs = npz['gt'][3]['images']` for this npz file. Maybe some operations in the model input preprocess avoid the harm of `nan` so that we have not found such issues during our training.

closed 2024-09-09 by huangjy-pku