arnold-benchmark/arnold #41: Problem in visualization

charlottejruby, 2024-09-06:
Thanks for your awesome work!

I am running Arnold using Docker. However, the Pick and Place task in Isaac Sim runs successfully within the Docker container, but the visualization does not work.

I think the visualization is not working due to the error shown below. Could you please help me resolve this issue? I would greatly appreciate it.

[image]

nikepupu, 2024-09-06:
Did you disable headless mode?

charlottejruby, 2024-09-09:
Yes, I tried to change the mode of eval.py
[image]

However, it doesn't work well.
Thanks for replying

huangjy-pku, 2024-09-09:
The error you mentioned does not matter. I guess the visualization should work well if you set `headless=False`.

charlottejruby, 2024-09-09:
[image]
It doesn't work well.

I keep getting errors like the one shown below. I've been toggling the headless mode option , which was originally set to False, then I changed it to True, and back to False again.

I don't think this is a headless mode issue. Could you provide a different solution? I am running it in a Docker setup

[image]

rgong-bdai, 2024-09-09:
Hi, can you show the part the program crashed? like the end of the log.

charlottejruby, 2024-09-10:
It seems that I encountered an error because I downloaded only part of the dataset.
Thank you for your help.

closed 2024-09-10 by charlottejruby