arnold-benchmark/arnold #43: ground truth action

charlottejruby, 2024-09-19:
First, thank you for your excellent work. 
While reading the paper, I had a question about how the ground truth action was generated. 
I understand that the key-point-based motion planner creates demonstrations. It seems that the motion planner was designed manually, and each task was broken down into sub-tasks, where the RMPflow controller was used to plan motion based on key points. Could you clarify what exactly is meant by "key-point" in this context?
Thank you

rgong-bdai, 2024-09-19:
depends on the task, keypoints denote the reference point computed based on mesh features (e.g, center of the handle for cabinet or drawers), or graspable parts for bottles or cups.

nikepupu, 2024-09-19:
depends on the task, keypoints denote the reference point computed based on mesh features (e.g, center of the handle for cabinet or drawers), or graspable parts for bottles or cups.

closed 2024-09-30 by huangjy-pku