arnold-benchmark/arnold #49: RuntimeError: CUDA out of memory

ChunkHsu, 2025-01-02:
When I use the following command  on 3090 GPU to train a task `/isaac-sim/python.sh train_peract.py task=pickup_object model=peract lang_encoder=clip mode=train batch_size=8 steps=10000`
then, I get an error `CUDA out of memory` when my batch_size=8:
```sh
root@ubuntu:~/arnold# /isaac-sim/python.sh train_peract.py task=pickup_object model=peract lang_encoder=clip mode=train batch_size=8 steps=10000
train_peract.py:155: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path='./configs', config_name='default')
/isaac-sim/kit/python/lib/python3.7/site-packages/hydra/_internal/hydra.py:127: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  configure_logging=with_log_configuration,
===> device:cuda
Loaded [100, 84, 86, 70, 119, 48, 36, 80] demos
Training 1000 steps, 623 demos, 8 batch_size
Error executing job with overrides: ['task=pickup_object', 'model=peract', 'lang_encoder=clip', 'mode=train', 'batch_size=8', 'steps=1000']
Traceback (most recent call last):
  File "train_peract.py", line 189, in main
    update_dict = agent.update(iteration, inp)
  File "/root/arnold/peract/agent.py", line 819, in update
    bounds)
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/arnold/peract/agent.py", line 602, in forward
    bounds)
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/arnold/peract/agent.py", line 308, in forward
    x = self_ff(x) + x
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/arnold/peract/network.py", line 648, in forward
    return self.fn(x, **kwargs)
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/arnold/peract/network.py", line 667, in forward
    return self.net(x)
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/container.py", line 117, in forward
    input = module(input)
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/arnold/peract/network.py", line 654, in forward
    return x * F.gelu(gates)
  File "/isaac-sim/kit/python/lib/python3.7/site-packages/torch/nn/functional.py", line 1383, in gelu
    return torch._C._nn.gelu(input)
RuntimeError: CUDA out of memory. Tried to allocate 32.00 MiB (GPU 0; 23.69 GiB total capacity; 22.27 GiB already allocated; 17.12 MiB free; 22.38 GiB reserved in total by PyTorch)

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
There was an error running python
```
How can I solve this problem?
(I have tried using batch_size=4 and encountered the same error(CUDA out of memory). I have also tried batch_size=2, but the best checkpoint accuracy obtained from the training is relatively low)

nikepupu, 2025-01-03:
How about gradient accumulation?

ChunkHsu, 2025-01-16:
> How about gradient accumulation?梯度累积怎么样？

ok, thanks

closed 2025-01-16 by ChunkHsu