arnold-benchmark/arnold #52: How can I generate a smooth trajectory video?

zero2vec, 2025-07-28:
Hello, thank you for your great work on this project.

I'm currently running the evaluation script in arnold on a headless server via SSH using a command like:

`bash python eval.py task=pickup_object model=peract lang_encoder=clip mode=eval use_gt=[0,0] visualize=0`

I would like to export a video of the rendered Isaac Sim simulation for each episode.
I tried collecting the images returned by the `render` function in `base_task.py` and stitching them into a video. However, the resulting video only contains about two frames in `pickup_object` task

Could you please advise how I can generate a smooth simulation video, like the demonstration shown on https://arnold-benchmark.github.io/?

Thank you in advance!

huangjy-pku, 2025-07-28:
Hi, thanks for your interest in our work! To record the simulation video, you can set `visualize=1` to turn on per-step rendering. If you have a screen rather than a headless server, you can use screen recording to record the simulation video. If you are using a headless server, there are two solutions to video recording: 1) adapt the task.recorder (e.g., [register_recorder](https://github.com/arnold-benchmark/arnold/blob/main/tasks/base_task.py#L416) and [try_record](https://github.com/arnold-benchmark/arnold/blob/main/tasks/base_task.py#L421)) to support recording observation; 2) hardcode manual saving functions around per-step simulation (e.g., [simulation_context.step](https://github.com/arnold-benchmark/arnold/blob/main/tasks/pickup_object.py#L182)).

zero2vec, 2025-07-28:
Thank you for your reply!

Following your suggestion to hardcode manual saving functions around each simulation step (e.g., simulation_context.step()), I configured the system to render and save an image at every step as shown below, and I also set `visualize=1`. However, I’m encountering an issue: once the episode begins, all of the saved images are identical. When I run the code above, `“Identical frames detected at timestep X”` is printed at every step, and the resulting video consists of the same image repeated. Could you please let me know if this behavior is expected?

```
            frames=[]
            ...
            simulation_context.step(render=render)
            self.time_step += 1

            current_frame = self.render()
            frames.append(current_frame)

            if len(frames) > 1:
                prev_frame = frames[-2]
                if np.array_equal(prev_frame['images'][0]['rgb'], current_frame['images'][0]['rgb']):
                    self.logger.warning(f"Identical frames detected at timestep {self.time_step}")

        if self.current_stage == self.num_stages:
            # stages exhausted, success check
            for _ in range(self.cfg.success_check_period):
                simulation_context.step(render=False)
                if self.checker.success:
                    self.is_success = 1
                    break
        
        return frames, self.is_success
```

closed 2025-07-28 by zero2vec