arnold-benchmark/arnold #54: How to selectively save a 3rd-person view only during rendering?

kelly062001, 2025-08-08:
Hello,

I'm trying to add a 3rd-person camera view to my simulation for visualization purposes, and I only want to save its frames when I'm actively rendering (e.g., for creating a video). 
I am saving observation frames at each timestep by calling current_frame = self.render() within my task's step function.

This method works perfectly for the five robot-mounted cameras that are defined in my _register_camera_path method:

```
# in BaseTask.py
def _register_camera_path(self):
    self.camera_paths = []
    
    robot_path = f'/World_{0}/franka'
    camera_paths = [ 
        f'{robot_path}/FrontCamera', 
        f'{robot_path}/BaseCamera',
        f'{robot_path}/LeftCamera',
        f'{robot_path}/panda_hand/GripperCameraBottom',
        f'{robot_path}/panda_hand/GripperCamera'
    ]
    # ...
```
These five cameras provide the agent with its first-person views. However, for visualization and debugging purposes, I would also like to obtain a 3rd-person view of the scene.
Could you please advise on the recommended way to add a new, 3rd-person camera to the simulation and retrieve its rendered frames?
For example, is the correct approach to programmatically create a new Camera prim at a desired location within my set_up_task method and then add its prim path to the list in _register_camera_path?

I'm looking for the standard procedure to include additional camera views beyond the ones defined on the robot asset itself.

Thank you for your help.

huangjy-pku, 2025-08-08:
Yes, I think you are right. The most straightforward way is to register a new camera and manually adjust its translation and rotation.

nikepupu, 2025-08-09:
You can also open the Franka USD we provided, and manually add a camera through isaac sim GUI.

closed 2025-08-11 by kelly062001